mprib/caliscope #480: Error while drawing item <pyqtgraph.opengl.items.GLScatterPlotItem.GLScatterPlotItem

yichuan1998, 2023-09-22:
Hi, thanks for your great job, this is a amazing tool. After multi camera calibration, I chose “HOLISTIC” and clicked “process”, finally got this error: "Error while drawing item <pyqtgraph.opengl.items.GLScatterPlotItem.GLScatterPlotItem".
[image]
what strange is that "HAND", "POSE" and "HOLISTIC_OPENSIM" work. Is there any problem? thanks!

mprib, 2023-09-22:
Thank you so much for reporting this and providing the log information! It is strange that Holistic_OpenSim works but Holistic doesn't so I want to get a better sense of the background of the problem.

If you click "Open Folder" it should open up the directory where the Holistic processing was done. It will be a subfolder of recording_2 and should include mp4 files that show the tracked points for each camera as well as an xy_HOLISTIC.csv file. The xyz file is missing, so it seems like some aspect of the pipeline leading to it failed (and without the xyz point data, pyqtgraph errors because it has nothing to draw).

I'm hoping to figure out where in the pipeline things failed. You can delete the HOLISTIC subfolder within recording 2 at which point you can re-run the processing the regenerate the logs.

Also, just to confirm: You were able to get HAND/POSE/HOLISTIC_OPENSIM to run on recording_2 specifically (not just recording_1)? I ask because several times I have tried to run processing with failed triangulation, only to discover that I did not actually have overlapping points in my camera's filed of view.

Thank you again for your feedback!

yichuan1998, 2023-09-23:
> Thank you so much for reporting this and providing the log information! It is strange that Holistic_OpenSim works but Holistic doesn't so I want to get a better sense of the background of the [...]

Mac, thanks for your immediate reply. I have tried recording many times of which "HOLISTIC" could not work but other three could. Also, pyxy3d would shut down after the log in terminal printed "Creating camera array projection matrices". As you said, recording_2 includes mp4 files and an xy_HOLISTIC.csv file while xyz file is missing.  Do you have any good idea? I will try it again with another computer and cameras
Additional text:
`mediapipe 0.10.1
opencv-contrib-python     4.8.0.76
numpy                     1.23.5
pyopengl                  3.1.7
pyqtgraph                 0.13.3
pyside6                   6.5.2
pyxy3d                    0.2.1
pytest                    7.4.2
python                    3.10.0 `

mprib, 2023-09-24:
Thank you for this information. As a double check I was able to do a fresh install on my laptop and process Holistic, so there may be something a little more complex going on than just code.

One thought: resources on your machine may be getting red-lined during the triangulation, causing the OS to shut the application down in self-defense. It is running scipy.svd() in a @jit function, which basically means it's going to push through lots of parallel processing. 

The Holistic face output has 468 points while pose (general body) has 33 and each hand has 21. Essentially, the face tracking in holistic is going to 10x your triangulation demands.

Because triangulation full Holistic was taking substantially longer than Pose or Hand, I created the Holistic_OpenSim tracker to filter the Holistic output to only include a few key points on the face that I could use for tracking the movement of the head. This significantly improved the triangulation processing, but sacrificed the rich face mesh.

I have 3 ideas:

1. **Expand the HolisticSim Output**: Are there keypoints on the face you would like to track that are not included in HolisticOpenSim? I could see about expanding the output of that tracker be less abbreviated but still more manageable than Holistic.
2. **Disable the @jit code and see if that stabilizes things**: Testing out this solution would be a bit more work on your end. Navigate to `your_project_directory\.venv\Lib\site-packages\pyxy3d\triangulate\triangulation.py` and comment out the @jit decorator lines at the top of the `def` lines in there. Save your changes and relaunch. This *might* reduce the intensity of the system utilization, though at the expense of speed.
3. **Process on a beefier machine**: it sounds like you are heading down this path, and if so I'm interested to learn how it goes.

Please let me know if any of this helps!

Best,

Mac

yichuan1998, 2023-09-25:
> Thank you for this information. As a double check I was able to do a fresh install on my laptop and process Holistic, so there may be something a little more complex going on than just code.
>
> One [...]

Mac, thanks for your so detailed reply and tips. I tracked the resources usage on my machine when processing HOLISTIC, founding that there were lots of RAM, CPU and GPU cache in idle state, which made me more confused. 
>
Resources on my machine:
`CPU: i9-13900k 64GB RAM RTX4070 12GB`
>
By your tip 2, after disabling @jit code, processing HOLISTIC worked eventually also with lots of resources unused. To do more test, I tested pyxy3d on machine 2 with i9-11900k, 80GB RAM and RTX2080TI 12GB, founding something unexpected. HOLISTIC Sometimes could work, sometimes did't. HAND, POSE, HOLISTIC_OPENSIM, only working few times, always failed with application shutting down in seconds.
[image]
[image]
>
 On machine 2, commenting out @jit code did't tackle the problem. It's so weird that in one recording file HOLISTIC could work while HNAD could't. What may lead to this phenomenon？“mp.solutions.hands.Hands” did't detect hands？
>
My operation process：
1. save charuco board config
2. single camera collect data and calibrate
3. multi-camera calibration
4. recording
5. post_process
>
Looking forward to your reply!
>
sincerely

mprib, 2023-10-09:
Apologies for the delay in my reply @yichuan1998. The situation you are describing is truly peculiar, particularly if sometimes it works and sometimes it doesn't. Makes pinning down a particular issue incredibly difficult.

Shot in the dark here: Is this variable behavior when processing the same underlying video files? I have sometimes experienced a frustrating lack of output only to realize that the subject was simply out of frame or the lighting was particularly bad where the subject was located. Still, this is not the kind of thing that would cause an application to outright crash. 

I am going to reflect on this and add additional logging to improve the picture of where things are breaking down. I'll reach out when a new version is pushed to PyPI, though it may be a day or two.

mprib, 2023-10-09:
Also, just to confirm: where mp4 files are being created in the processing subfolder, do you see the tracked landmarks being drawn on the frames? These should also show up as points in the xy.csv files. I just want to confirm if that stage of the processing is happening successfully.

yichuan1998, 2023-10-10:
Mac, I'm sure that the subject is in the middle of the frame and lighting is good for each recording. Everytime it broke up after processing, I checked the subfolder and found mediapipe working well, landmarks drawn on the frames. xy.csv files was not none. By your new version is pushed, I will test again. if  there is any new discovery, I will feedback to you.

mprib, 2023-10-10:
@yichuan1998, I have updated a new version to PyPI (0.2.4). This includes a couple of fixes compared to the version from a couple weeks ago. For my own notekeeping I will add this (please feel free to ignore this minutia): it removes an errant import of a development dependency that made its way into a module that sometimes gets called during post-processing. This might be responsible for the failure of some of the triangulations, though if sometimes things were working, then that is definitely strange.

One final shot in the dark based on an issue I had when doing a fresh install and complete workflow:

If the rotation and translation vectors for each camera are not set, then the triangulation cannot happen. I did encounter a strange occurrence of believing that these had been set and saved, only to discover that the config.toml file within the recording directory did not have this information saved. If this is the issue driving the lack of triangulation on your end, then I'll know I need to drill down on why this data is sometimes not getting locked down prior to recording.

On the recordings that you are unable to triangulate, could you confirm if all cameras have the rotation/translation data saved in config.toml?

---

I wanted to also underscore my gratitude for you taking the time to provide this feedback. It really is invaluable to improving the stability of this package. Thank you.

yichuan1998, 2023-10-11:
I rechecked the config.toml in recording directory and root directory, confirming rotation/translation data was saved. 
[image]
->
After updating the package to 0.2.4, I repeted testing several times of calibration and post processing this morning. Now everything works well. The problem of program crash has not occurred again on machine2 mentioned before. 
->
Thanks for your improvement to this. To be honest, I was looking for reliable and accurate calibration tools in past few weeks. The pyxy3d is the most efficient、convenient tool I have ever used. I will continuously use this package. 
->
Sincerely

mprib, 2023-10-11:
Thank you for the kind words. They are certainly encouraging. If there are features that you would like to see integrated at some point, please just raise an issue or open up a conversation in the discussions.

closed 2023-10-11 by mprib