mprib/caliscope #481: Failure to launch on MacOS 14.1 Sonoma

mprib, 2023-10-09:


Corey-PT, 2023-10-09:
Here's the error output on my Macbook pro:

Traceback (most recent call last):
  File "../opt/anaconda3/envs/pyxy3d/bin/pyxy3d", line 5, in <module>
    from pyxy3d.__main__ import launch
  File "../opt/anaconda3/envs/pyxy3d/lib/python3.10/site-packages/pyxy3d/__init__.py", line 16, in <module>
    __app_dir__ = Path(os.getenv("LOCALAPPDATA"), __package_name__)
  File "../opt/anaconda3/envs/pyxy3d/lib/python3.10/pathlib.py", line 960, in __new__
    self = cls._from_parts(args)
  File "../opt/anaconda3/envs/pyxy3d/lib/python3.10/pathlib.py", line 594, in _from_parts
    drv, root, parts = self._parse_args(args)
  File "../opt/anaconda3/envs/pyxy3d/lib/python3.10/pathlib.py", line 578, in _parse_args
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType

mprib, 2023-10-10:
Hi @Corey-PT ! Based on the error message you provided, it appears that at initialization, the operating system is being misidentified as windows (using os.name == "nt"). I've revised the code to use a different method of checking (platform.system). This is working on macOS 13, the most recent version of the OS I can readily test. 

This new code has been pushed to PyPI (v0.2.4). If you uninstall and reinstall into your .venv via pip, hopefully this should work. Please let me know if this doesn't fix the problem!

Thank you,

Mac

mprib, 2023-10-13:
Thanks for the update @Corey-PT. It may take some time for me to block out to drill down into this. In the meantime I'll update the Readme with the issues on sonoma. I'm leaving a link here for my own reference to start exploring potential solutions:

https://stackoverflow.com/questions/68659865/cannot-pip-install-mediapipe-on-macos-m1

rgov, 2023-11-07:
At first glance, it works OK on on Sonoma 14.0 for me. I installed from `HEAD` into a Python 3.10 `venv` virtual environment. Python was installed with Homebrew, not Anaconda. Note you must use Python 3.10 since Numba doesn't support newer releases. No issue installing any of the dependencies including MediaPipe.

[image]

mprib, 2023-11-07:
@rgov,

Thank you for the MacOS and Numba information. I will pin the python version to 3.10 with that in mind, and close the issue with confirmation that it launches.

General comment since you are taking a look at things: I wanted to keep you in the loop that I'm in the midst of a fairly large refactor to switch the calibration to be based off of pre-recorded footage. The intention is to allow easier integration with systems based around something other than webcams (which is what I'm learning more individuals want). I'm doing my best to not touch the currently existing functionality until I have a full scale replacement, but I wanted to keep you aware that things may get "delicate" while this work is in process.

Thanks again for your interest in the project and the recent PR. If you have any questions or requests please just let me know. 
Mac

closed 2023-11-07 by mprib

rgov, 2023-11-07:
@mprib Thanks. I very much want to use pre-recorded footage for calibration, so I hope this is available for testing soon.