mprib/caliscope #561: New/Open Project doesn't work anymore

rgov, 2023-12-04:
macOS 14, pyxy3d 61065143e92da658883e716fffe8e2e6b56878f4.

When I select "New/Open Project" in the menu, nothing happens.

mprib, 2023-12-04:
Hi @rgov,

I'm in the final stages of getting the pre-recorded workflow organized in the GUI (literally working on it right now). In a day or two I might have things ready to test and find problems. I'll reply again to this issue and @ you when it's ready to go. 
 
Mac

rgov, 2023-12-04:
Cool. Fwiw, this was broken in 7056781 due to removal of the call to `connect_menu_actions()`.

closed 2023-12-05 by mprib

mprib, 2023-12-05:
@rgov 

The current `main` (as well as the 0.3.0 version on PyPI have been updated such that the launching of the workspace should be operable now.

There are some known issues with how the capture volume widget is handled (this is where you can inspect that solution from the extrinsic parameter calibration and set the origin), and I hope to clear that up in the next day or two.

I'm going to out together a sample workflow with some old webcam recorded footage and maybe post a quick video walk through.

In the meantime, many things should be working. In particular, the first stage of the calibration (intrinsic camera setup) should be working. Within workspace/calibration/intrinsic, you can have individual files named port_1.mp4, port_2.mp4, etc. These should be videos with the charuco board being moved through the field of view so that individual frame grabs can be selected to provide the input for the intrinsic camera calibration. 

Just wanted to point that out as something that I believe is fully functional and can be played around with. I'd be very interested in your thoughts on the user experience with that portion of the calibration.

mprib, 2023-12-05:
@rgov...just a note that I was likely premature in suggesting it might be ready to look at. I'm going through the workflow from scratch with some stuff and finding lots of little GUI-hanging issues that I need to clean up. I'll check back in when I have things stable...

mprib, 2023-12-07:
@rgov,

Just wanted to mention that the current version is (to my knowledge) usable. I've thrown a couple patches up tonight to fix things that I saw during this demo video, but in general the workflow functions:

- https://www.youtube.com/watch?v=XDfuQ_7m1oQ

I did a walk-through of the required underlying file structure here:

- https://youtu.be/dvosFrFVDsU

ASIDE: The RMSE of reprojections in this walk through are definitely larger than they would be with a higher quality board/cameras. I was just trying to throw some demo footage together with what was easily accessible.