mprib/caliscope #583: Issues with

rgov, 2023-12-10:
I created a new project and set up the file structure as described. The config.toml file was populated like so for each camera:

```
matrix = "null"
distortions = "null"
```

It seems that these are being loaded not as `None` but as 0-dimension Numpy arrays. Therefore, when  `pyxy3d.cameras.camera_array.CameraData.get_display_data()` does these checks, it tries to access invalid indices of the array:

```python
        if self.matrix is not None:
            fx, fy = self.matrix[0, 0], self.matrix[1, 1]
            cx, cy = self.matrix[0, 2], self.matrix[1, 2]
        ...
        if self.distortions is not None:
            k1, k2, p1, p2, k3 = self.distortions.ravel()[:5]
```

You could change them to `if self.distortions is not None and len(self.distortions.shape) >= 5` etc. but a better solution might be to have it load the values as None not empty arrays.

rgov, 2023-12-10:
Later on I also received:

```
File "pyxy3d/calibration/stereocalibrator.py", line 237, in stereo_calibrate
    camera_matrix_A = np.array(camera_matrix_A, dtype=float)
ValueError: could not convert string to float: 'null'
```

mprib, 2023-12-10:
Thank you for flagging this. In the last couple of days I switched over to rtoml from toml (there was a reason for this....loading in the point_estimates in the new workflow was bogging down the GUI). It may be that rtoml handles storing None differently than toml, so I need to clean up how that is loaded in. I'll take a look today.

closed 2023-12-10 by mprib

mprib, 2023-12-10:
This issue should be resolved now with the most recent patch. I'm going to turn my attention to automating the initial population of the Intrinsic calibration boards. Squaring that away will likely take a bit longer than this issue, but hopefully I can have something working in a couple days.