mprib/caliscope #607: Can I use python 3.7 on pyxy3d?

JohnNan123, 2024-01-16:
Hi, I tried to implement pyxy3d with our FLIR camera system, which requires PySpin to connect to the camera instead of cv2.VideoCapture. However, the PySpin version I use only supports python 3.7. I want to know if pyxy3d will still function in a python 3.7 environment. 

Thanks

mprib, 2024-01-16:
Hi @JohnNan123 ,

Pyxy3D is now set up to work with pre-recorded footage so that the video acquisition can be decoupled from the video processing. My hope is that this will allow a larger variety of camera solutions. If you can get your data in the [standard format](https://mprib.github.io/pyxy3d/project_setup/) then it should be possible to process it apart from any specific camera requirements (so hopefully its own Python interpreter is fine).

This is a quick and dirty sample project that hopefully gives a more concrete example of the file layout requirements: https://utexas-my.sharepoint.com/:f:/g/personal/priblede_my_utexas_edu/EliBf-eSuxJMpltXfANlS4oB35S9muKtpCxg4VMKX9rt2Q?e=skyiTm

I am not familiar with PySpin but am definitely interested in learning more about it. Side Note: I have moved the recording (built on OpenCV) to its own package that I'm still in the midst of fleshing out: https://github.com/mprib/multiwebcam

I'm guessing PySpin already has some synchronized recording functionality available along with hardware trigger integration, but just thought I'd point out that companion project as a potential resource that could potentially be adapted to non-OpenCV capture objects.

mprib, 2024-01-16:
And I'm realizing that I have not directly addressed your question about 3.7. I believe that there may be some type hinting that would clash with 3.7, but if you run into that issue and are willing to sort it out, I'm totalling willing to accept a PR to make the main project work with more python versions.

Another potential issue is Mediapipe, which can be a particularly finicky dependency and I can't really speak to how that might behave.

JohnNan123, 2024-01-16:
Hi @mprib 
Thank you for your reply. The fact is we used to have pyxy3d working in our camera as same as cv2.VideoCapture uses a wrapper called EasyPySpin, which is available from Git Hub, under Python 3.10. We still want to keep the live-streaming of the calibration feature as it is easy to use. The only problem is the Python 3.7 version, which has some syntax changes (like match/cases and __slot__ at @dataclass ), which stops it from running.

mprib, 2024-01-17:
@JohnNan123 

I'm glad to know that the live stream version has been useful. The feedback that I've previously received was universally of the opinion that people wanted to be able to integrate with pre-recorded footage. Trying to maintain both approaches is tricky (so far development is primarily just me), so I've opted to only support the pre-recorded workflow. I hope you understand and will consider giving the pre-recorded workflow a try if this is otherwise a useful tool.

One advantage of this new workflow is that it can capture many more frames because it is not running charuco searches on each of the cameras which slows down the processing. I hope that the added robustness from more data will offset some of the conveniences of the real-time feedback. 

My plan is to get [multiwebcam](https://github.com/mprib/multiwebcam) saving out everything in the appropriate file structure so that it would be a matter of "click <record calibration> in multiwebcam" and then once files are saved out "click <process calibration footage> in pyxy3d". 

---

And thank you for pointing out PySpin and EasyPySpin to me. These are projects I previously didn't know about and it is super relevant to me. I'm sorry that I may not be able to prioritize maintaining backward compatibility with 3.7 given my time constraints. If there is any feedback you have about the project aside from the python compatibility I'm very interested in hearing it. And if you feel like contributing at some point, PRs are welcome!

mprib, 2024-01-20:
Going to go ahead and close this one. Worth noting that not even the Python Software Foundation supports Python 3.7 anymore: https://devguide.python.org/versions/

My hope is that PySpin and its underlying packages will get brought forward to python versions that have not yet reached end-of-life.

closed 2024-01-20 by mprib