mprib/caliscope #617: External Video

fanpeeps, 2024-02-05:
How to import video captured with GoPro or other source?

mprib, 2024-02-07:
The project should be structured according to the file layout detailed here: https://mprib.github.io/caliscope/project_setup/

This an example project folder: https://utexas-my.sharepoint.com/:f:/g/personal/priblede_my_utexas_edu/EliBf-eSuxJMpltXfANlS4oB35S9muKtpCxg4VMKX9rt2Q?e=Ecwi62

If you save that folder to your computer and then launch caliscope you should be able to open that project folder and get to a workspace which is the starting point for the processing. The walk-through of that process starts here:  https://mprib.github.io/caliscope/workspace/

If this doesn't answer your questions please just let me know. In the next couple weeks I'll probably post a video that recapitulates the info in the walk-through in a more digestible way... I'm keeping this open (and renaming) to make sure that stays on the top of my to-do

mprib, 2024-07-23:
appears to have addressed user"s question, so closing out.

closed 2024-07-23 by mprib