mprib/caliscope #618: blender addon

carlosedubarreto, 2024-02-09:
Hello, it came to me that you were making this very interesting solution for markerless mocap. (a friend at discord told me about)

I'm building blender addons with some interesting codes that I find on github. Most of them are free and I make a few of them premium versions to try to support myself and help to pay the bills.

I dont have much free time, but maybe it would be cool to try to join forces and make a cool addon with the work you've been doing.

what do you think?

And seeing your will to make a free cool markerless mocap tool, I'm thinking of making my tool that is a bit similar to yours (but its a blender addon) to remove its price and make it avilable for free too.

anyway, I have a gumroad page where most of the addons are free or have a full free working version 
https://carlosedubarreto.gumroad.com

and I have a patreon site where I try to use to support myself while trying to do cool things for the blender community
https://www.patreon.com/cebstudios

Let me know if you'd like some help on the blender part.

Cheers

mprib, 2024-02-10:
@carlosedubarreto,

Thank you for reaching out about this! My motion capture background is rooted in biomechanics and the workflow of animators is sufficiently different that I generally feel overwhelmed trying to understand what may be useful to them. Collaborating on a project sounds like a wonderful idea! 

The blender rig animation currently at the top of the README was first put together in an ad-hoc way, but I split it out into a rough attempt at a blender add-on in a separate project called [Rigmarole](https://github.com/mprib/rigmarole). 

The README of that project attempts to give a sense of the project.  I don't know if my code for that is terribly comprehensible but hopefully it could serve as a starting point for a conversation about this.

I would be very interested in your thoughts.

Thank you again!

Mac

mprib, 2024-04-15:
Haven't heard anything back on this one...

carlosedubarreto, 2024-04-15:
@mprib im so sorry, i ended up getting into 4 freelance  projects while having a dayjob.
These last 2 months were crazy.

Sorry for not answering earlier, it was not possible to do much.

I think and hope that in 2 months things get lighter on my side

mprib, 2024-04-24:
@carlosedubarreto,

Sorry for my delay in responding! I'm reopening this just to keep it more "alive" when I get a chance to review issues. Totally understand the time constraints and I look forward to being able to make progress on something together in the future.

Best,

Mac