mprib/caliscope #622: Bus Error on Apple Silicon on clicking "Calibrate Capture Volume"

calcoloergosum, 2024-04-10:
Thank you for great tool.

- `np.linalg.inv`  is called in non-main thread. (`caliscope.cameras.camera_array_initializer.get_inverted_stereopair`).
- On Apple silicon, certain numpy function calls do not work outside main thread (!).
- The following lines fix this issue. (credit to [this answer in stackoverflow](https://stackoverflow.com/a/75317069))
```
export MKL_NUM_THREADS=1
export NUMEXPR_NUM_THREADS=1
export OMP_NUM_THREADS=1
```
- Thus, this issue is caused by a bug in external library

It may be nice to get this issue up for future reference.

mprib, 2024-04-15:
Thank you for this information! I've updated the README with a link back to this issue. I'm running Windows at the moment so I really appreciate you being able to provide this help.

mprib, 2024-04-15:
Closing now that this is made explicit on the README

closed 2024-04-15 by mprib