mprib/caliscope #624: Out of memory error during extrinsic "Calibrate Capture Volume" calibration

Gnarlywhale, 2024-04-18:
Error encountered while running the "Calibrate Capture Volume" step (intrinsic calibration had no issues). Happened with both an older laptop (2.4 GHz i5, 8 GB RAM) and a more powerful desktop (3.7 GHz i5, 32 GB RAM), both running Windows 10.

Source clips were from eight 1920x1080, 30 fps, hardware-synched cameras (300MB each). 

Looks like it's likely an optimization issue with FFmpeg/Opencv:
`[ WARN:1@549.271] global cap_ffmpeg.cpp:441 cv_capture_retrieve FFmpeg: Exception is raised: OpenCV(4.9.0-pre) /build/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 6220800 bytes in function 'OutOfMemoryError'`

I'll take a closer look at the source to see if there's any low-hanging fruit for a fix, but just wanted to flag it in the interim.

Gnarlywhale, 2024-04-24:
Quick note, closing the preview window dramatically reduced the resource usage and I was able to complete the calibration. The CHARUCO board video files were also all present in the extrinsic calibration folder, as expected.

mprib, 2024-04-24:
@Gnarlywhale ,

Thank you for bringing this up. One thought I have relates to the way that frames build up in a queue when they are waiting to e written to a video file. With higher resolutions and more cameras I would expect those queues build up deeper and deeper. When the single python process is trying to also render the video playback it may cause the stack to seriously back up and choke out the memory. 

One response to this may be to just have a check box to elect to view the real time progress so that you don't have to be mindful to close out of it when resource use is pushing the limits. I'm keeping this issue up and hope to have time to take a look at this. Realisitically, I'm heading into my comprehensive exam next month so coding tasks are falling on the back burner right now and I hope to pick them up in the summer.

Thank you for bringing this to my attention! I hope that caliscope is working for your use case despite these hiccups!

Mac

mprib, 2024-07-29:
@Gnarlywhale,

It's been awhile, but I wanted to point out that I've made some changes discussed in #626 that I hope may address the issue that you described. My relevant response on that issue is here:

https://github.com/mprib/caliscope/issues/626#issuecomment-2256933166

Regards,

Mac

closed 2024-09-13 by mprib