mprib/caliscope #625: Not connecting to python

maxbatt22, 2024-05-05:
Hello there,

I've been trying to follow the instructions for Mac, although I continue to get errors that Python doesn't exist. I am sure I have used Python 3.1 for other apps, so is it possible I am somehow referencing the wrong path? 

/opt/homebrew/bin/python3 -m venv .venv

[image]

mprib, 2024-05-05:
Hi @maxbatt22,

You may not have python 3.10 installed, or if you do then you need to specifically reference python 3.10 and not just python 3, which will default to some version and maybe not the one you want. Note that python 3.10 is not the same as python 3.1 (semantic versioning looks like a decimal system but it is not).

When you run `python3` from the terminal do you get an output like:

```
Python 3.10.10 (tags/v3.10.10:aad5f6a, Feb  7 2023, 17:20:36) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

Regards,

Mac

maxbatt22, 2024-05-05:
Hello Mac, 

Thank you so much! I am planning on using this tool for the next episode of my show, AI MASHUPS.

I think you're right. Python is ON my system, but every time I type `python --version`  into Terminal, it launches Xcode to install more software (then seemingly doesn't install it).

But then when I type in your command, it is almost identical

```
Python 3.9.6 (default, Jun 28 2021, 19:24:41) 
[Clang 12.0.5 (clang-1205.0.22.9)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
```

Previously, when I tried to find the path of Python, the word "Homebrew" was in the path. Perhaps Homebrew installed the wrong version for Caliscope to use?

mprib, 2024-05-06:
Hi @maxbatt22,

I'm primarily on Windows these days and occasionally Ubuntu as needed, so I don't have good specific guidance for navigating MacOS issues. It seems like you are having trouble setting up a python 3.10 virtual environment.

Do you have something on your filesystem named python3.10? If so, can you run the install code from the README?:

```

# Create a virtual environment named 'venv' using Python 3.10
/path/to/python3.10 -m venv .venv

```
If not, then you may need to install 3.10:  https://www.python.org/downloads/release/python-3100/

Regards,

Mac

mprib, 2024-07-23:
closing as this was likely a python environment issue.

closed 2024-07-23 by mprib