mprib/caliscope #641: Scale Input Boxes to Improve visability of numbers

mprib, 2024-09-15:
Report from DavidPagnon that the spin boxes were not large enough to display text:

[image]

Documenting target appearance of GUI:

[image]

Also, for reference, this is how the charuco should update when paper size changes:

[image]

davidpagnon, 2024-09-17:
Hi @mprib,

Thank you for taking into account all these suggestions!

I'm still not sure I understand: 
Let's say you set the board to have 4 rows x 3 columns. Let's say you specified that the square size is 5.4 cm. Then the Charuco board should be 21.6 x 16.2 cm². 
If you specified that the board size should be 25 x 20 cm², shouldn't you have margins on all sides? More specifically, 1.7 cm on top and bottom, and 1.9 on the right and left sides?

[image]

davidpagnon, 2024-09-17:
Actually, I think I got it: if we did not manage to print the board properly, this can be fixed before proceeding to the next stage by correcting the actual square edge length. Is this it?

mprib, 2024-09-17:
> this can be fixed before proceeding to the next stage by correcting the actual square edge length. Is this it?

Yes, it is this exactly. The scale of the world is set by the actual size of the squares. In practice there is generally a small deviation of the final printed page from the "perfect" intended measurement. 

This number ultimately just goes into the charuco board definition in OpenCV: https://docs.opencv.org/3.4/d0/d3c/classcv_1_1aruco_1_1CharucoBoard.html

mprib, 2024-10-22:
Note that in https://github.com/mprib/caliscope/issues/652, it is reported that the width is getting capped at 100, and a0 format will exceed that. Need to expand viable inputs here as well.

mprib, 2024-10-23:
@davidpagnon,

I've just pushed some changes and updated to PyPI. I believe that the charuco widget will now not constrain the widget width in such a way that it clips the text. 

If you have a moment. could you confirm if the change worked? I can't duplicate the original issue on my machines (I'm guessing because of my resolution?)

Thank you!

Mac

davidpagnon, 2024-10-24:
Hi, this is great on most text fields, aside from a few of them:

[image]

[image]

mprib, 2024-10-24:
@davidpagnon,

Thank you for looking at this. I really appreciate it.

I've just pushed changes to `main` that attempt to set the text box based on an assessment of the font metrics for the font being used, along with some extra padding. Hopefully that will do that trick.... 🤞

Mac

closed 2025-03-17 by mprib