mprib/caliscope #646: Caliscope calibration support in Pose2Sim

davidpagnon, 2024-09-22:
Not an issue, just keeping you informed that Caliscope calibration is now natively supported in [Pose2Sim](https://github.com/perfanalytics/pose2sim) and in the [Pose2Sim_Blender](https://github.com/davidpagnon/Pose2Sim_Blender) addon :)

https://github.com/davidpagnon/Pose2Sim_Blender/assets/54667644/a2cfb75d-a2d4-471a-b6f8-8f1ee999a619

mprib, 2024-09-25:
WOW! That is awesome, @davidpagnon!

Your blender add-on is exactly the kind of thing that I'd thought would provide an excellent way to examine the output of a motion trial (projecting the image onto the camera and being able to snap the view to the camera). But you actually IMPLEMENTED it. Fantastic to see that in action. 

Thank you for keeping me in the loop. I'm about to go into a black hole as I prepare for my dissertation proposal next month, but I hope we can keep in touch on these projects for more bandwidth becomes available.

davidpagnon, 2024-09-25:
You're welcome, thanks for the appreciation! On the other hand, with rigmarole you rigged the whole thing, and I haven't found time to look into it so that could be a good source of inspiration.

I really need to stop working on the nitty-gritty of everything as well, because I've got other urgent projects; so there is no rush at all. And good luck (or "bon courage" as we say in French) for your dissertation! 💪✍️
And it would be a pleasure to work together in the future if we have the occasion :)