mprib/caliscope #650: Support output to OpenPose JSON format

mprib, 2024-10-07:
OpenPose provides a standard JSON format for its output:

https://cmu-perceptual-computing-lab.github.io/openpose/web/html/doc/md_doc_02_output.html

Rather than attempting to create some new style of representation, it may be best to default to a previously existing format that has survived some tire-kicking.

Note that OpenPose links to a format for representing the camera matrix, but the link is currently 404'd: https://cmu-perceptual-computing-lab.github.io/openpose/web/html/doc/advanced/calibration_module.md

davidpagnon, 2024-10-07:
Two other standards are:
- DeepLabCut (one single CSV file per sequence), which actually two standards. One for single, and one for multiple animals
- Coco/MMPose (one json file)

I can send you samples of OpenPose, deeplabcut, and Coco files if you are interested.

mprib, 2024-10-07:
@davidpagnon,

Thank you. I am interested so if those are easily accessible it would be awesome to have some references. 

Mac

mprib, 2024-10-08:
@davidpagnon,

This is incredibly helpful to have as a reference, thank you!  It's great to see how others approach the problem of storing this data. An ecosystem of open source tools that can interoperate to facilitate mocap is near and dear to my heart, so this is much appreciated. 

Mac

mprib, 2026-02-28:
Tracked internally. Low priority — CSV and TRC cover current output needs.

closed 2026-02-28 by mprib