mprib/caliscope #660: RMSE of each port pair looks good, but the overall error is large

wanghlsara, 2024-10-29:
I mentioned in #659 that I performed extrinsic calibration in a narrow and long rectangular area. I copied some lines related to rmse from calibration.log. Please forgive me that these lines may not be continuous because I ignored the information unrelated to rmse. What confuses me is that the rmse of each port pair looks good, but the overall error is large. What could be the reason? Is it because there are few cameras with overlapping views in the rectangular area? Or is the ratio of FILTERED_FRACTION=0.025 too small? Or is the number of bundle adjustment stages too small?  What aspects should I investigate next? I hope to get your advice.

```
     INFO|         caliscope.configurator| 205|  charuco param are: {'columns': 4, 'rows': 3, 'board_height': 35.43, 'board_width': 44.88, 'dictionary': 'DICT_4X4_100', 'units': 'inch', 'aruco_scale': 0.75, 'square_size_overide_cm': 26.6, 'inverted': False, 'legacy_pattern': False}:
     INFO|  caliscope.calibration.charuco| 106|  Creating charuco with square length of 0.266
     INFO|  caliscope.calibration.charuco| 116|  Setting legacy pattern of board to False
     INFO|                       __main__|  79|  Creating RecordedStreamPool
     INFO|         caliscope.configurator| 152|  Adding camera 1 to calibrated camera array...
     INFO|         caliscope.configurator| 166|  Camera successfully added at port 1
     INFO|         caliscope.configurator| 152|  Adding camera 10 to calibrated camera array...
     INFO|         caliscope.configurator| 178|  Camera array successfully created and being passed back to caller
     INFO| caliscope.calibration.stereocalibrator|  39|  666:[(1, 2), (1, 3), (1, 4), (1, 5), (1, 6), (1, 7), (1, 8), (1, 9), (1, 10), (1, 11), (1, 12), (1, 13), (1, 14), (1, 15), (1, 16), (1, 17), (1, 18), (1, 19), (1, 20), (1, 21), (1, 22), (1, 23), (1, 24), (1, 25), (1, 26), (1, 27), (1, 28), (1, 29), (1, 30), (1, 31), (1, 32), (1, 33), (1, 34), (1, 35), (1, 36), (1, 37), (2, 3), (2, 4), (2, 5), (2, 6), (2, 7), (2, 8), (2, 9), (2, 10), (2, 11), (2, 12), (2, 13), (2, 14), (2, 15), (2, 16), (2, 17), (2, 18), (2, 19), (2, 20), (2, 21), (2, 22), (2, 23), (2, 24), (2, 25), (2, 26), (2, 27), (2, 28), (2, 29), (2, 30), (2, 31), (2, 32), (2, 33), (2, 34), (2, 35), (2, 36), (2, 37), (3, 4), (3, 5), (3, 6), (3, 7), (3, 8), (3, 9), (3, 10), (3, 11), (3, 12), (3, 13), (3, 14), (3, 15), (3, 16), (3, 17), (3, 18), (3, 19), (3, 20), (3, 21), (3, 22), (3, 23), (3, 24), (3, 25), (3, 26), (3, 27), (3, 28), (3, 29), (3, 30), (3, 31), (3, 32), (3, 33), (3, 34), (3, 35), (3, 36), (3, 37), (4, 5), (4, 6), (4, 7), (4, 8), (4, 9), (4, 10), (4, 11), (4, 12), (4, 13), (4, 14), (4, 15), (4, 16), (4, 17), (4, 18), (4, 19), (4, 20), (4, 21), (4, 22), (4, 23), (4, 24), (4, 25), (4, 26), (4, 27), (4, 28), (4, 29), (4, 30), (4, 31), (4, 32), (4, 33), (4, 34), (4, 35), (4, 36), (4, 37), (5, 6), (5, 7), (5, 8), (5, 9), (5, 10), (5, 11), (5, 12), (5, 13), (5, 14), (5, 15), (5, 16), (5, 17), (5, 18), (5, 19), (5, 20), (5, 21), (5, 22), (5, 23), (5, 24), (5, 25), (5, 26), (5, 27), (5, 28), (5, 29), (5, 30), (5, 31), (5, 32), (5, 33), (5, 34), (5, 35), (5, 36), (5, 37), (6, 7), (6, 8), (6, 9), (6, 10), (6, 11), (6, 12), (6, 13), (6, 14), (6, 15), (6, 16), (6, 17), (6, 18), (6, 19), (6, 20), (6, 21), (6, 22), (6, 23), (6, 24), (6, 25), (6, 26), (6, 27), (6, 28), (6, 29), (6, 30), (6, 31), (6, 32), (6, 33), (6, 34), (6, 35), (6, 36), (6, 37), (7, 8), (7, 9), (7, 10), (7, 11), (7, 12), (7, 13), (7, 14), (7, 15), (7, 16), (7, 17), (7, 18), (7, 19), (7, 20), (7, 21), (7, 22), (7, 23), (7, 24), (7, 25), (7, 26), (7, 27), (7, 28), (7, 29), (7, 30), (7, 31), (7, 32), (7, 33), (7, 34), (7, 35), (7, 36), (7, 37), (8, 9), (8, 10), (8, 11), (8, 12), (8, 13), (8, 14), (8, 15), (8, 16), (8, 17), (8, 18), (8, 19), (8, 20), (8, 21), (8, 22), (8, 23), (8, 24), (8, 25), (8, 26), (8, 27), (8, 28), (8, 29), (8, 30), (8, 31), (8, 32), (8, 33), (8, 34), (8, 35), (8, 36), (8, 37), (9, 10), (9, 11), (9, 12), (9, 13), (9, 14), (9, 15), (9, 16), (9, 17), (9, 18), (9, 19), (9, 20), (9, 21), (9, 22), (9, 23), (9, 24), (9, 25), (9, 26), (9, 27), (9, 28), (9, 29), (9, 30), (9, 31), (9, 32), (9, 33), (9, 34), (9, 35), (9, 36), (9, 37), (10, 11), (10, 12), (10, 13), (10, 14), (10, 15), (10, 16), (10, 17), (10, 18), (10, 19), (10, 20), (10, 21), (10, 22), (10, 23), (10, 24), (10, 25), (10, 26), (10, 27), (10, 28), (10, 29), (10, 30), (10, 31), (10, 32), (10, 33), (10, 34), (10, 35), (10, 36), (10, 37), (11, 12), (11, 13), (11, 14), (11, 15), (11, 16), (11, 17), (11, 18), (11, 19), (11, 20), (11, 21), (11, 22), (11, 23), (11, 24), (11, 25), (11, 26), (11, 27), (11, 28), (11, 29), (11, 30), (11, 31), (11, 32), (11, 33), (11, 34), (11, 35), (11, 36), (11, 37), (12, 13), (12, 14), (12, 15), (12, 16), (12, 17), (12, 18), (12, 19), (12, 20), (12, 21), (12, 22), (12, 23), (12, 24), (12, 25), (12, 26), (12, 27), (12, 28), (12, 29), (12, 30), (12, 31), (12, 32), (12, 33), (12, 34), (12, 35), (12, 36), (12, 37), (13, 14), (13, 15), (13, 16), (13, 17), (13, 18), (13, 19), (13, 20), (13, 21), (13, 22), (13, 23), (13, 24), (13, 25), (13, 26), (13, 27), (13, 28), (13, 29), (13, 30), (13, 31), (13, 32), (13, 33), (13, 34), (13, 35), (13, 36), (13, 37), (14, 15), (14, 16), (14, 17), (14, 18), (14, 19), (14, 20), (14, 21), (14, 22), (14, 23), (14, 24), (14, 25), (14, 26), (14, 27), (14, 28), (14, 29), (14, 30), (14, 31), (14, 32), (14, 33), (14, 34), (14, 35), (14, 36), (14, 37), (15, 16), (15, 17), (15, 18), (15, 19), (15, 20), (15, 21), (15, 22), (15, 23), (15, 24), (15, 25), (15, 26), (15, 27), (15, 28), (15, 29), (15, 30), (15, 31), (15, 32), (15, 33), (15, 34), (15, 35), (15, 36), (15, 37), (16, 17), (16, 18), (16, 19), (16, 20), (16, 21), (16, 22), (16, 23), (16, 24), (16, 25), (16, 26), (16, 27), (16, 28), (16, 29), (16, 30), (16, 31), (16, 32), (16, 33), (16, 34), (16, 35), (16, 36), (16, 37), (17, 18), (17, 19), (17, 20), (17, 21), (17, 22), (17, 23), (17, 24), (17, 25), (17, 26), (17, 27), (17, 28), (17, 29), (17, 30), (17, 31), (17, 32), (17, 33), (17, 34), (17, 35), (17, 36), (17, 37), (18, 19), (18, 20), (18, 21), (18, 22), (18, 23), (18, 24), (18, 25), (18, 26), (18, 27), (18, 28), (18, 29), (18, 30), (18, 31), (18, 32), (18, 33), (18, 34), (18, 35), (18, 36), (18, 37), (19, 20), (19, 21), (19, 22), (19, 23), (19, 24), (19, 25), (19, 26), (19, 27), (19, 28), (19, 29), (19, 30), (19, 31), (19, 32), (19, 33), (19, 34), (19, 35), (19, 36), (19, 37), (20, 21), (20, 22), (20, 23), (20, 24), (20, 25), (20, 26), (20, 27), (20, 28), (20, 29), (20, 30), (20, 31), (20, 32), (20, 33), (20, 34), (20, 35), (20, 36), (20, 37), (21, 22), (21, 23), (21, 24), (21, 25), (21, 26), (21, 27), (21, 28), (21, 29), (21, 30), (21, 31), (21, 32), (21, 33), (21, 34), (21, 35), (21, 36), (21, 37), (22, 23), (22, 24), (22, 25), (22, 26), (22, 27), (22, 28), (22, 29), (22, 30), (22, 31), (22, 32), (22, 33), (22, 34), (22, 35), (22, 36), (22, 37), (23, 24), (23, 25), (23, 26), (23, 27), (23, 28), (23, 29), (23, 30), (23, 31), (23, 32), (23, 33), (23, 34), (23, 35), (23, 36), (23, 37), (24, 25), (24, 26), (24, 27), (24, 28), (24, 29), (24, 30), (24, 31), (24, 32), (24, 33), (24, 34), (24, 35), (24, 36), (24, 37), (25, 26), (25, 27), (25, 28), (25, 29), (25, 30), (25, 31), (25, 32), (25, 33), (25, 34), (25, 35), (25, 36), (25, 37), (26, 27), (26, 28), (26, 29), (26, 30), (26, 31), (26, 32), (26, 33), (26, 34), (26, 35), (26, 36), (26, 37), (27, 28), (27, 29), (27, 30), (27, 31), (27, 32), (27, 33), (27, 34), (27, 35), (27, 36), (27, 37), (28, 29), (28, 30), (28, 31), (28, 32), (28, 33), (28, 34), (28, 35), (28, 36), (28, 37), (29, 30), (29, 31), (29, 32), (29, 33), (29, 34), (29, 35), (29, 36), (29, 37), (30, 31), (30, 32), (30, 33), (30, 34), (30, 35), (30, 36), (30, 37), (31, 32), (31, 33), (31, 34), (31, 35), (31, 36), (31, 37), (32, 33), (32, 34), (32, 35), (32, 36), (32, 37), (33, 34), (33, 35), (33, 36), (33, 37), (34, 35), (34, 36), (34, 37), (35, 36), (35, 37), (36, 37)]
     INFO| caliscope.calibration.stereocalibrator|  40|  ===========
     INFO| caliscope.calibration.stereocalibrator| 175|  Deleting previous stereocalibrations from config
     INFO| caliscope.calibration.stereocalibrator| 181|  Beginning stereocalibration of pairs [(1, 2), (1, 3), (1, 4), (1, 5), (1, 6), (1, 7), (1, 8), (1, 9), (1, 10), (1, 11), (1, 12), (1, 13), (1, 14), (1, 15), (1, 16), (1, 17), (1, 18), (1, 19), (1, 20), (1, 21), (1, 22), (1, 23), (1, 24), (1, 25), (1, 26), (1, 27), (1, 28), (1, 29 [... cut ...]
     INFO| caliscope.calibration.capture_volume.capture_volume| 110|  Following bundle adjustment (stage 2), RMSE is: 337.69850592993237
     INFO|         caliscope.configurator| 264|  Saving camera array....
     INFO|         caliscope.configurator| 317|  Saving point estimates to toml...
     INFO|                       __main__| 121|  Asserting that RMSE decreased with optimization at overall...
     INFO|                       __main__| 121|  Asserting that RMSE decreased with optimization at 1...
```

wanghlsara, 2024-10-29:
In order to increase the number of iterations of optimization, I used following code in test_calibration.py

```
 quality_controller = QualityController(capture_volume, charuco)
    for _ in range(5):
        logger.info("Filtering out worst fitting point estimates")
        quality_controller.filter_point_estimates(0.25)
        quality_controller.capture_volume.optimize()
```

and got python exception:
 ```
 File "C:\WORK\caliscope-main\tests\test_calibration.py", line 140, in test_post_monocalibration
    quality_controller.capture_volume.optimize()
  File "C:\WORK\caliscope-main\caliscope\calibration\capture_volume\capture_volume.py", line 93, in optimize
    self.least_sq_result = least_squares(
  File "C:\Github\00.caliscope\.venv\lib\site-packages\scipy\optimize\_lsq\least_squares.py", line 832, in least_squares
    f0 = fun_wrapped(x0)
  File "C:\Github\00.caliscope\.venv\lib\site-packages\scipy\optimize\_lsq\least_squares.py", line 830, in fun_wrapped
    return np.atleast_1d(fun(x, *args, **kwargs))
  File "C:\WORK\caliscope-main\caliscope\calibration\capture_volume\capture_volume.py", line 162, in xy_reprojection_error
    points_3d = current_param_estimates[capture_volume.point_estimates.n_cameras * CAMERA_PARAM_COUNT :].reshape(
ValueError: cannot reshape array of size 93213 into shape (31065,3)
```
I wonder how to fix this exception in my code? Looking forward to your suggestions.

mprib, 2024-10-29:
Just tacking on to this to record some thoughts.... my expectation is that the whole camera array is kinda weakly joined together in the sense that you have many stereopairs that just don't have common views. And you have many common views where they are only linked at the edge of the view where distortion has the greatest impact, and you have lots of disparity in distance to the camera such that you are combining near and far viewpoints which I think would create an interesting tension between trying to minimize the pixel distance error while guessing at the 3D location.

It would take me some time to really get back into the depths of the bundle adjustment and how points are filtered out. I think that what is happening is that you have so many high error 2D points in the bundle adjustment that when they get filtered out they are losing complete points in the 3D point estimates. This is something that is driven by the extremes mentioned in the paragraph above. A resolvable issue, but I'd at least need access to your intermediate data (config.toml and charuco_xy) and it would take some time.

1. What are the camera resolutions? 
2. What is the size and configuration of the board?
3. I don't mean to pry, but just out of curiosity, what domain is this in?

As another shot in the dark, the lines of code I'd suggested you remove in #659 are there to allow use of a double sided board sandwiched between glass with a mirror image board printed on the back. I'd originally created that for the specific use case of cameras that are mounted far from each other on either end of a hallway. It would allow more links between cameras that otherwise couldn't be linked because they can't face the same surface of the board at the same time. So you could try creating one of these mirror boards, uncommenting that code, and running the frame rate at something much less (I feel like 5 fps would be a good starting point).

wanghlsara, 2024-10-29:
> Those RMSE are quite large. I will say that as I try to wrap my head around what you are doing I am quite curious about what the actual setup of this is. You have 37 cameras lining a hallway. How are they configured to record synchronized frames?

I recorded videos using some industrial cameras with hardware-level frame synchronization capabilities

wanghlsara, 2024-10-29:
1.What are the camera resolutions?
1440x1080

 2.What is the size and configuration of the board?
[charuco]
columns = 4
rows = 3
board_height = 35.43
board_width = 44.88
dictionary = "DICT_4X4_100"
units = "inch"
aruco_scale = 0.75
square_size_overide_cm = 26.6
inverted = false
legacy_pattern = false

 3.I don't mean to pry, but just out of curiosity, what domain is this in?
sports analysis

>  A resolvable issue, but I'd at least need access to your intermediate data (config.toml and charuco_xy) and it would take some time.

Attachments are the data files, Thanks for your assistance!
[testdata.zip](https://github.com/user-attachments/files/17560160/testdata.zip)

mprib, 2025-11-01:
Closing this out. I definitely appreciate the issue being raised as it gives a lot to think about. I continue to suspect that this behavior is a consequence of the hardware setup/camera configuration and fundamentally driven by the 'thin connections" between cameras. An interesting use case and I see the value in it, but there are some very real constraints in the data that are hard to overcome.