mprib/caliscope #664: Using chessboard instead of charuco?

seyed-mohammadreza-mousavi, 2024-12-11:
Hello and thanks for your project. Is it possible we use chessboard instead of charuco board for calibrating?

mprib, 2024-12-14:
Hi @seyed-mohammadreza-mousavi,

The Charuco is pretty central to the calibration process as it helps to resolve the otherwise indeterminate rotation of the calibration board, so I'm sorry to say that it's not currently something I plan to support. 

I'm curious why you would prefer to use a chessboard. If I could better understand the motivation for a potential refactor, it might help me to prioritize it.

Thank you,

Mac

mprib, 2025-01-04:
Closing as no details provided to argue in favor of chessboard pattern.

closed 2025-01-04 by mprib