mprib/caliscope #667: Installation error - mediapipe version

kgyb, 2025-01-03:
First of all thank you for all you efforts making this project available to everyone!

Im a newbie in this field and just starting out so excuse me for that. I'm trying to install calliscope but get an error when doing so:
**D:\3D Apps\Caliscope\caliscope-main>pip install .
Processing d:\3d apps\caliscope\caliscope-main
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting PyOpenGL<4.0.0,>=3.1.6 (from caliscope==0.5.3)
  Using cached PyOpenGL-3.1.7-py3-none-any.whl.metadata (3.2 kB)
INFO: pip is looking at multiple versions of caliscope to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement mediapipe==0.10.1 (from caliscope) (from versions: 0.10.5, 0.10.7, 0.10.8, 0.10.9, 0.10.10, 0.10.11, 0.10.13, 0.10.14, 0.10.18, 0.10.20)
ERROR: No matching distribution found for mediapipe==0.10.1** 
This is what I got after trying to install it manually (downloading it from github and installing the file) as I've already tried installing it the recommended way using simply the **pip install caliscope** command and that didn't wok either. Using the recommended method it said it to be installed properly but when I tried to run the program using the **caliscope** command it gave an error saying there is no such command.

After all that I tried to do it manually as you can see above, but that didn't work. I even tried installing mediapipe manually too but I don't seem to find the proper version listed anywhere on the official mediapipe repo. Can I maybe use another version of mediapipe and edit the requirements in the installer files for it to accept a newer version?

Im on windows 10 btw using python 3.11

mprib, 2025-01-03:
@kgyb,

Thank you for raising an issue! 

Given the output about "multiple versions of caliscope" my hunch is that this may have something to do with the active python environment. I recently changed the mediapipe dependencies because mediapipe had deprecated some older versions which were no longer available.

I did a fresh editable install and it appears to be working. These are the basic steps I took (in Powershell):

```powershell
# out of an abundance of caution, point directly to the executable  python you want to copy into the .venv
# note that this step can take a moment while it copies the python instance
# if you already have caliscope installed in that instance, it might create some issues, i'm not sure... so perhaps uninstall there if it is
& "C:\Users\Mac Prible\AppData\Local\Programs\Python\Python311\python.exe -m venv .venv"

# activate the virtual environment. A green (.venv) should appear on the left in powershell
.\.venv\Scripts\activate

# from here you can launch caliscope
caliscope
```

Activating the virtual environment will look like this:

[image]

As a check of which python is running, you can do the following:

[image]

I hope this addresses the underlying issue and provides some clarification about python. In practice it's a bit easier to use some tools to manage your virtual environments (like [`uv`](https://www.youtube.com/watch?v=qh98qOND6MI)), but then it's always helpful to know how to do it the long way to make sure you understand where all these things are happening.

Please let me know if this doesn't address the issue or you have any other questions!

Mac

kgyb, 2025-01-03:
Thank you! Editing the **_pyproject.toml_** file to require a new mediapipe version (as you did in the new update) solved the issue. This issue had came up just before you've pushed to update and then christmas came and was busy haha so I didnt see it. Thank you again for the fast response! You rule man! 😀

mprib, 2025-01-04:
Glad to hear it worked out!

closed 2025-01-04 by mprib