mprib/caliscope #671: Support for multiple calibration boards

charisoudis, 2025-03-13:
Hi and thanks for the nice work!

We are working in a semi-circular configation with some cameras being placed at a high altitude. It would be nice for us to able to use more calibration boards so that at least one is visible from all/most of the cameras, something like the image below:

[image]

Could this be possible using caliscope? Any guidance on where we should focus our attention for extending it?

Best,
Thanos

mprib, 2025-03-13:
@charisoudis, 

Thank you for your question! I based the calibration process on my experience with a traditional vicon system where a calibration wand is waved around the intended capture volume and placed on a landmark to set the origin. Is there a technical limitation that means you can't get the desired coverage by moving the board slowly through space?

Even if only a few cameras at a time have a single view of the board, all of the overlapping views get rolled up in the final adjustment process.

Perhaps related, you can increase redundancy by printing a mirror image of the board on the back (an extremely thick board may present some challenges there). Doing this, you could have the camera on the upper left and the camera on the upper right localize the same points in the space at the same instant which may improve the quality of the calibration.

Please let me know your thoughts on the above. I'm going to start taking a look at your other issue and will reply on that thread...

Best,

Mac

mprib, 2026-02-28:
Tracked internally. Worth noting: pairwise stereo bootstrapping already handles the case where no single board position is visible to all cameras — each camera pair only needs shared visibility of one board.

closed 2026-02-28 by mprib