mprib/caliscope #672: AttributeError: 'Synchronizer' object has no attribute 'port'. Did you mean: 'ports'?

charisoudis, 2025-03-13:
Any idea why this happens in the latest release?

Here are some logs:

```bash
 ...
 caliscope.cameras.synchronizer|Placing new synched frames with index 1900 
 caliscope.cameras.synchronizer|Placing new synched frames with index 1900 
 caliscope.recording.video_recorder|Size of unsaved frames on the recording queue is 1000 
 caliscope.controller|Waiting for 2D tracked points to populate at C:\Users\Thanos\PycharmProjects\capturestudio\out\Captures\l12_s2x2_c12_synced_@13_03_2025_18_16_03\caliscope\calibration\extrinsic\CHARUCO\xy_CHARUCO.csv 
 caliscope.recording.recorded_stream|Ending recorded playback at port 2 
 caliscope.controller|Waiting for 2D tracked points to populate at C:\Users\Thanos\PycharmProjects\capturestudio\out\Captures\l12_s2x2_c12_synced_@13_03_2025_18_16_03\caliscope\calibration\extrinsic\CHARUCO\xy_CHARUCO.csv 
 caliscope.recording.video_recorder|Size of unsaved frames on the recording queue is 1025 
 caliscope.cameras.synchronizer|Placing new synched frames with index 2000 
 caliscope.cameras.synchronizer|Placing new synched frames with index 2000 
 caliscope.recording.recorded_stream|Ending recorded playback at port 11 
 caliscope.cameras.synchronizer|End of frames at port 1 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 2 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 3 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 4 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 5 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 6 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 7 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 8 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 9 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 10 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 11 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 1 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 2 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 3 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 4 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 5 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 6 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 7 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 8 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 9 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 10 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 11 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 1 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 2 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 3 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 4 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 5 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 6 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 7 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 8 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 9 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 10 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 11 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 1 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 2 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 3 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 4 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 5 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 6 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 7 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 8 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 9 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 10 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 11 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 1 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 2 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 3 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 4 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 5 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 6 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 7 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 8 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 9 detected; ending synchronization 
 caliscope.cameras.synchronizer|End of frames at port 10 detected; ending synchronization 
[... 252 lines cut ...]
    return op.apply().__finalize__(self, method="apply")
  File "C:\Users\Thanos\miniconda3\lib\site-packages\pandas\core\apply.py", line 764, in apply
    return self.apply_standard()
  File "C:\Users\Thanos\miniconda3\lib\site-packages\pandas\core\apply.py", line 891, in apply_standard
    results, res_index = self.apply_series_generator()
  File "C:\Users\Thanos\miniconda3\lib\site-packages\pandas\core\apply.py", line 907, in apply_series_generator
    results[i] = self.f(v)
  File "C:\Users\Thanos\miniconda3\lib\site-packages\pandas\core\apply.py", line 142, in f
    return func(x, *args, **kwargs)
  File "C:\Users\Thanos\miniconda3\lib\site-packages\caliscope\calibration\stereocalibrator.py", line 58, in get_coverage_region
    label = row[port]
  File "C:\Users\Thanos\miniconda3\lib\site-packages\pandas\core\series.py", line 981, in __getitem__
    return self._get_value(key)
  File "C:\Users\Thanos\miniconda3\lib\site-packages\pandas\core\series.py", line 1089, in _get_value
    loc = self.index.get_loc(label)
  File "C:\Users\Thanos\miniconda3\lib\site-packages\pandas\core\indexes\base.py", line 3804, in get_loc
    raise KeyError(key) from err
KeyError: 11
 caliscope.recording.video_recorder|About to emit `all frames saved` signal 
 caliscope.gui.frame_emitters.frame_dictionary_emitter|End of playback signalled by synchronizer 
Traceback (most recent call last):
  File "C:\Users\Thanos\miniconda3\lib\site-packages\caliscope\gui\frame_emitters\frame_dictionary_emitter.py", line 80, in run
    logger.info(f"Thread loop within frame emitter at port {self.synchronizer.port} successfully ended")
AttributeError: 'Synchronizer' object has no attribute 'port'. Did you mean: 'ports'?
```

charisoudis, 2025-03-14:
Hi @mprib and thanks for your prompt reply.

Regarding the former bug, it resulted in the GUI not proceeding to store the `point_estimates.toml` and thus required us to restart the capture volume calibration. I had to manually update the line in `frame_dictionary_emitter.py`:
```
logger.info(f"Thread loop within frame emitter at port {self.synchronizer.port} successfully ended")
```
to
```
logger.info(f"Thread loop within frame emitter at port {self.synchronizer.ports[0]} successfully ended")
```

Attached you may also find the requested files, though we see that there are overlapping views in port11 with most of the other cameras (I had to rename the `toml` file to `toml.txt` for gh to accept it).

Best,
Thanos

[config.toml.txt](https://github.com/user-attachments/files/19245106/config.toml.txt)

[xy_CHARUCO.csv](https://github.com/user-attachments/files/19245121/xy_CHARUCO.csv)

mprib, 2025-03-15:
@charisoudis,

Thank you for the data files! Just to clarify, **were you able to proceed successfully after changing that line of code?** In my own testing, I can see that my logs get that error as well, though it doesn't stop the calibration. The window with the synchronized frames is intended to automatically close at the same time that print statement is logged, so I don't think that is the primary source of the issue. I suspect a couple of potential issues and I hope to narrow it down. 

In good news, I was able to create the following calibration from your provided files. Please let me know if this looks right to you:

https://www.youtube.com/watch?v=aBWQyVlWGMk

In my code to create these point estimates, I had to avoid a refining stage of the processing pipeline after it seemed to hang for an extended period of time. You should be able to avoid this stage by commenting out the following lines of `controller.py` (beginning line 373):

```
            self.quality_controller = QualityController(self.capture_volume, self.charuco)

            logger.info(f"Removing the worst fitting {FILTERED_FRACTION*100} percent of points from the model")
            self.quality_controller.filter_point_estimates(FILTERED_FRACTION)
            self.capture_volume.optimize()
```

I suspect your calibration data may be unearthing an edge case in how this filtering is done that I need to resolve, so I'm grateful for you raising this issue. This stage of the processing is not critical to the results, it just refines the rotation/translation estimates after filtering out the worst fitting 2.5% of points (the rationale is that these poor fits may be due to intrinsic parameter error or point identification error and therefore may be undermining the calibration). 

Testing with your data has also underscored how time intensive the stereocalibration can be and is something that I may now try to improve.

Please let me know if you are able to progress in the calibration workflow by fixing the logger error or by commenting out the QualityController issue. I will keep hacking away on my end.

Thank you!

Mac

mprib, 2025-03-19:
@charisoudis,

After duplicating issues with the calibration process using your data, I was able to pin down where some processing bottlenecks were causing the calibration to grind to a halt (this stage coincided with when the call to `synchronizer.port` threw an error) . Following the changes merged in today, the complete pipeline, including filtering out the 2.5% of poor fits, appears to proceed quickly. Please note that you may need to flick "Reload Workspace" to see the results of the calibration in the Capture Volume tab.

I am grateful to you for bringing this to my attention and for providing the data that enabled me to identify where the calibration could hang. If you have any additional issues using caliscope, please don't hesitate to reach out. I will leave this open for a couple more days and would be additionally grateful if you could confirm for me that the latest release resolves this issue. 

Thank you,

Mac

closed 2025-10-29 by mprib