mprib/caliscope #673: how to calibrate checkerboard images?

DEVECLOVER, 2025-03-17:
Can the checkerboard images be calibrated by this great work, or can only ChArUco images be calibrated? The calibration example shows processing of calibration videos corresponding to three cameras. If it is a set of corresponding images collected by the three cameras, how should it be processed?

Thanks very much!!!

mprib, 2025-03-17:
Hello @DEVECLOVER,

Caliscope is only set to work with charuco boards, not checkerboards. If I'm understanding your second question, you are asking how to process a set of photos with each frame in its own file? I think the easiest thing to do would be to convert those to mp4 using opencv and then just process the mp4 as a standard video file. 
 
Regards,

Mac

closed 2025-03-17 by mprib

DEVECLOVER, 2025-03-21:
> Hello [@DEVECLOVER](https://github.com/DEVECLOVER),
>
> Caliscope is only set to work with charuco boards, not checkerboards. If I'm understanding your second question, you are asking how to process a [...]

OK, Thank you!