mprib/caliscope #681: Numba core error when testing out triangulation via python API

aruna-ram, 2025-03-26:
Hi Mac! 
When I tested out the triangulation from csv file like this  
```python 
from pathlib import Path 
from caliscope.triangulate.triangulation import triangulate_from_files

# Define paths
config_path = Path("./config.toml")
xy_path = Path("./xypoints_caliscope.csv")
output_path = Path("./xyz_data.csv")

# Perform triangulation
xyz_data = triangulate_from_files(
    config_path=config_path,
    xy_path=xy_path, 
    output_path=output_path
)

print(f"Triangulated {len(xyz_data)} points")
```

, I encounter the following error:

```bash 
Traceback (most recent call last):
  File "./triangulation_caliscope.py", line 13, in <module>
    xyz_data = triangulate_from_files(
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "./caliscope/caliscope/triangulate/triangulation.py", line 271, in triangulate_from_files
    xyz_data = triangulate_xy(xy_data, camera_array)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "./caliscope/caliscope/triangulate/triangulation.py", line 139, in triangulate_xy
    point_id_xyz, points_xyz = triangulate_sync_index(projection_matrices, port, point_ids, raw_xy)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "./caliscope_env/lib/python3.11/site-packages/numba/core/dispatcher.py", line 468, in _compile_for_args
    error_rewrite(e, 'typing')
  File "./caliscope_env/lib/python3.11/site-packages/numba/core/dispatcher.py", line 409, in error_rewrite
    raise e.with_traceback(None)
numba.core.errors.TypingError: Failed in nopython mode pipeline (step: nopython frontend)
non-precise type array(pyobject, 1d, C)
During: typing of argument at ./caliscope/caliscope/triangulate/triangulation.py (66)

File "caliscope/caliscope/triangulate/triangulation.py", line 66:
def unique_with_counts(arr):
    <source elided>

@jit(nopython=True, parallel=True, cache=True)
```

I haven't worked with Numba before, but it could be something about defining of types in the lists made in  `triangulation_sync_index` function within `triangulation.py` - but you probably know better!

Let me know if you need more info :)

mprib, 2025-03-26:
Thank you, @aruna-ram. Could you share your files and I'll see if I can duplicate/debug on my end?

aruna-ram, 2025-03-26:
Here they are:

[xypoints_caliscope.csv](https://github.com/user-attachments/files/19469172/xypoints_caliscope.csv)

I cannot seem to attach .toml files, so here is my config.toml file

```toml
camera_count = 2
creation_date = 2025-03-24T15:22:21.529169
fps_sync_stream_processing = 100
save_tracked_points_video = true

[cam_1]
port = 1
rotation_count = 0
error = 0.535
exposure = "null"
grid_count = 27
size = [1920, 1236]
matrix = [[859.4325716845894, 0.0, 966.1572178228552], [0.0, 875.2304538512118, 679.1302952071342], [0.0, 0.0, 1.0]]
distortions = [-0.40721015948037287, 0.32485356534918036, -0.013322058767006177, 0.00039675652420144447, -0.1715186259176031]
translation = [0.0, 0.0, 0.0]
rotation = [1.5707963267948966, 0.0, 0.0]

[cam_2]
port = 2
rotation_count = 0
error = 0.587
exposure = "null"
grid_count = 32
size = [1920, 1236]
matrix = [[1156.425719821471, 0.0, 1036.1002284104907], [0.0, 1155.2740499548336, 579.2888042502539], [0.0, 0.0, 1.0]]
distortions = [-0.38128917010849556, 0.2010690628371262, -0.0007483556676477311, -0.0006787772103330546, -0.04288944310068429]
translation = [-0.31889512318088864, -0.019785351585145845, 0.3611161605106117]
rotation = [2.0208857348098404, 2.1685039669767088, -0.29263165081307424]

[capture_volume]
stage = 2
origin_sync_index = 181

[charuco]
columns = 5
rows = 7
board_height = 11.0
board_width = 8.5
dictionary = "DICT_4X4_50"
units = "inch"
aruco_scale = 0.75
square_size_overide_cm = 4.0
inverted = false
legacy_pattern = false

```

mprib, 2025-03-26:
@aruna-ram,

Thank you for the input data. The `point_id` must be an integer value and it looks like this was what was causing the issue. I was able to generate the xyz data after changing "feeder" --> `1`:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/MACPRI~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/MACPRI~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#467886" vlink="#96607D">

sync_index | point_id | x_coord | y_coord | z_coord
-- | -- | -- | -- | --
1 | 1 | 0.149235 | 0.349785 | 0.013764

</body>

</html>

Please let me know if this resolves things on your end!

Mac

aruna-ram, 2025-03-26:
ah my bad! thanks, I get no errors now :) (apart from a NumbaTypeSafetyWarning about casting from int64 to in32)

mprib, 2025-03-26:
Glad to hear it! I will close this out.

If you end up with some data that you would feel comfortable making public I would be grateful to see how it panned out and to share it as a demonstration of how caliscope can be used (but if you prefer not, I completely understand).

Best,

Mac

closed 2025-03-26 by mprib

aruna-ram, 2025-03-26:
Sure, I'll be doing experiments with hawkmoths where I track their approaches to a flower if this would suit? It might take a couple more months to provide you with the dataset.

mprib, 2025-03-26:
Sounds awesome! Better integrating with DLC for use with animal studies is an important future direction for this project. I work in human biomechanics/gait and most folks will say: just tape markers on the person and put them in the $80k vicon system. Not always a simple thing with non-human subjects...