mprib/caliscope #684: Better Calibration

crocis, 2025-03-28:
I am currently getting calibrations with a reprejection error larger than 1 pixel with 4K videos. How can I get a lower reprojection error? Is there a way to control the quality of the calibration through the hyperparameters of the optimization?

mprib, 2025-03-28:
some ideas:

1. improve the quality of the single camera calibrations by [providing better input](https://mprib.github.io/caliscope/intrinsic_calibration/#practical-recording-guidelines)

2. improve the quality of the single camera calibrations by [adding a more refined criteria parameter](https://github.com/mprib/caliscope/blob/c896d05696ec6dd2afb2e88503fb9c410009130b/caliscope/calibration/intrinsic_calibrator.py#L217):

```
# play with modifying default criteria here
criteria = (cv2.TERM_CRITERIA_EPS + cv2.TERM_CRITERIA_MAX_ITER, 30, 0.001)

self.error, self.mtx, self.dist, self.rvecs, self.tvecs = cv2.calibrateCamera(
    self.calibration_obj_loc,
    self.calibration_img_loc,
    (width, height),
    None,
    None,
    criteria=criteria
)
```

3. If you have really good single camera parameters, you could also get more aggressive with filtering out more poor fits during the bundle adjustment by increasing the [filtered fraction](https://github.com/mprib/caliscope/blob/c896d05696ec6dd2afb2e88503fb9c410009130b/caliscope/controller.py#L33). The rationale for this is that overfitting single camera parameters might lead to issues when calibrating relative position of cameras. You could toss out more poor fits with the idea that it might be due to a bad distortion estimate or something like that.

4. Improve the quality of the charuco tracking by playing with the [criteria for the corner detection](https://github.com/mprib/caliscope/blob/c896d05696ec6dd2afb2e88503fb9c410009130b/caliscope/trackers/charuco_tracker.py#L25)

Please let me know if any of these help improve the reprojection error. I am genuinely interested in learning if there are ways to improve the defaults and potentially building out a straightforward way to adjust these parameters via the config file.

mprib, 2025-03-28:
@crocis,

I would also add:

5. play with the parameters of the bundle adjustment:

```

mprib, 2025-04-09:
No response or follow up from OP

closed 2025-04-09 by mprib