mprib/caliscope #686: JSON file for rigmarole use is missing

kgyb, 2025-05-13:
Hello!

Sorry for reaching out here aswell but I opened an issue on the rigmarole repo but didn't get an answer yet, I didn't know if it's because you're really busy or you dont get notifications to the repo anymore. I wanted to import the mocap data into blender for later use in other 3d software but im missing the JSON file with the body size predictions and even when I tried to import with the default sized one you provided it didn't work properly :( and I don't know how I could fix this. For more information you can see the issue ticket I opened on the other repo, but if you feel so, feel free to answer here. Thanks in advance!

mprib, 2025-05-13:
Hello! Sorry I didn't see the Rigmarole issue. I've been a bit distracted trying to cross t's and dot i's on my dissertation, so my brain is a bit of swiss cheese these days.....

kgyb, 2025-05-14:
Thank you I didn't notice that! It works with the simple holistic model now. I only have one last question, is there any quality difference between the holistic and simple holistic model except the lack of face tracking? Is there any additional filtering on the holistic that the simple doesn't have? Im asking because when I tested the addon it only worked with the .csv that came out of the simple model and not the full holistic. When I tried to import those then I only got the movement from the root bone(?) which moved the skeleton in one piece around. Sorry for bothering you again with my questions haha, I will definitely give credit to you if I ever release my project im planning to use this on.

mprib, 2025-05-14:
> , is there any quality difference between the holistic and simple
holistic model except the lack of face tracking?

Nope, that's the only difference. The face points were bogging down the
processing, and the pose model provides low-quality wrist tracking, so I
tried to create something that was more "medium" in terms of quality and
output size. And I believe it's only set up to work with the simple model,
so it wouldn't surprise me if the holistic data didn't quite map over.
Thank you for your questions and good luck with your project!

Mac

closed 2025-08-21 by mprib