mprib/caliscope #688: Extrinsic calibration fails due to "no shared points" even though there some

felixfrank, 2025-06-18:
I need some help with extrinsics calibration for a setup with 4 cameras. The cameras have overlap like this (1,2), (2,3) and (3,4).
I recorded a video of me moving around a 800x600mm high precision charuco calibration board and the intrinsic calibration works well.
However when I get to calibrating the extrinsics, caliscope fails to calibrate the (3,4) stereo pair.
This is the log output:

```shell
     INFO| caliscope.calibration.stereocalibrator| 242|  Deleting previous stereocalibrations from config
     INFO| caliscope.calibration.stereocalibrator| 248|  Beginning stereocalibration of pairs [(1, 2), (1, 3), (1, 4), (2, 3), (2, 4), (3, 4)]
     INFO| caliscope.calibration.stereocalibrator| 170|  Assembling 10 shared boards for pair (1, 2)
     INFO| caliscope.calibration.stereocalibrator| 316|  RMSE of reprojection for pair (1, 2) is 1.0746374045284361
     INFO| caliscope.calibration.stereocalibrator| 152|  For pair (1, 3) there are no shared points
     INFO| caliscope.calibration.stereocalibrator| 319|  No stereocalibration produced for pair (1, 3)
     INFO| caliscope.calibration.stereocalibrator| 152|  For pair (1, 4) there are no shared points
     INFO| caliscope.calibration.stereocalibrator| 319|  No stereocalibration produced for pair (1, 4)
     INFO| caliscope.calibration.stereocalibrator| 170|  Assembling 10 shared boards for pair (2, 3)
     INFO| caliscope.calibration.stereocalibrator| 316|  RMSE of reprojection for pair (2, 3) is 1.0947902449310738
     INFO| caliscope.calibration.stereocalibrator| 152|  For pair (2, 4) there are no shared points
     INFO| caliscope.calibration.stereocalibrator| 319|  No stereocalibration produced for pair (2, 4)
     INFO| caliscope.calibration.stereocalibrator| 152|  For pair (3, 4) there are no shared points
     INFO| caliscope.calibration.stereocalibrator| 319|  No stereocalibration produced for pair (3, 4)
     INFO| caliscope.calibration.stereocalibrator| 266|  Direct stereocalibration complete for all pairs for which data is available
```
However, when I closely inspect the xy_CHARUCO.csv I find quite a few frames that have shared points for the (3,4) pair. Or maybe I am interpreting the csv in the wrong way?

Here is an excerpt of the csv:
[xy_CHARUCO_part.csv](https://github.com/user-attachments/files/20805270/xy_CHARUCO_part.csv)

The entire csv is around 65MB. Is it a problem because it is too large?

mprib, 2025-06-19:
@felixfrank,

Sorry you are having this difficulty. Looking at your csv file, I agree that it has the data that should allow the calibration to happen.

I don't anticipate that the file size is causing the issue. Could you share the complete csv along with your `config.toml` so that I can step through the processing on my end to pinpoint what may be going on?

Thank you,

Mac

felixfrank, 2025-06-19:
Hey Mac,

Thank you for your help. 
[Here is the full csv](https://drive.google.com/file/d/1EyPJwKbc4osb8e69HDAFUL7rVAxMrxle/view?usp=sharing). 
Here is the config.toml:
```toml
camera_count = 4
creation_date = 2025-06-18T16:11:40.621336
fps_sync_stream_processing = 100
save_tracked_points_video = true

[cam_1]
port = 1
rotation_count = 0
error = 1.711
translation = "null"
rotation = "null"
exposure = "null"
grid_count = 99
size = [1280, 800]
matrix = [[653.5125274226923, 0.0, 627.8705474392154], [0.0, 645.5179102638297, 401.60689526610895], [0.0, 0.0, 1.0]]
distortions = [-0.3058112505192529, 0.09376398027137665, -0.0011700361180094522, 0.001881781827522227, -0.012564899012438185]

[cam_2]
port = 2
rotation_count = 0
error = 1.522
translation = "null"
rotation = "null"
exposure = "null"
grid_count = 99
size = [1280, 800]
matrix = [[604.1833863708309, 0.0, 632.6712938846276], [0.0, 605.1453457744788, 416.41584740636296], [0.0, 0.0, 1.0]]
distortions = [-0.25774767198663023, 0.06129483546870621, -0.0008654940339133301, -0.0002779037113538956, -0.006110033068220707]

[cam_3]
port = 3
rotation_count = 0
error = 6.211
translation = "null"
rotation = "null"
exposure = "null"
grid_count = 99
size = [1280, 800]
matrix = [[654.7318539166104, 0.0, 621.687634616621], [0.0, 650.0720557054946, 402.82211402328073], [0.0, 0.0, 1.0]]
distortions = [-0.3121850279658758, 0.09603586976968095, -0.0014883124083917608, -0.000356348209254505, -0.012565554528596864]

[cam_4]
port = 4
rotation_count = 0
error = 1.217
translation = "null"
rotation = "null"
exposure = "null"
grid_count = 99
size = [1280, 800]
matrix = [[593.5026994968729, 0.0, 622.5452515831788], [0.0, 591.5977577083357, 414.4912203898179], [0.0, 0.0, 1.0]]
distortions = [-0.26192239112571547, 0.06464680879780084, -0.0009440166009447545, -0.00035483086530404493, -0.006741906458063692]

[charuco]
columns = 12
rows = 9
board_height = 60.0
board_width = 80.0
dictionary = "DICT_5X5_250"
units = "cm"
aruco_scale = 0.75
square_size_overide_cm = 6.0
inverted = false
legacy_pattern = false

[stereo_1_2]
RMSE = 1.0746374045284361
rotation = [[0.5463002787228205, 0.009312816085990995, -0.8375376271690229], [-0.037370228558142836, 0.9992134445337517, -0.013264926705692312], [0.8367553235473801, 0.0385456057099085, 0.5462186053180315]]
translation = [[-0.06168009020422238], [-0.03170364014627701], [-0.09058801912854968]]

[stereo_2_3]
RMSE = 1.0947902449310738
rotation = [[0.5410890859010563, 0.015734669246334366, -0.840818066707934], [-0.012199901055221452, 0.9998666018281908, 0.010860062743325459], [0.8408767826102718, 0.004381635796665293, 0.5412088670141532]]
translation = [[-0.06821903887669489], [-0.008192691506793304], [-0.0037829476402517505]]

```

felixfrank, 2025-06-19:
I can look at it as well. Do you have a hunch on what might be wrong? Maybe I can narrow down the code I look at.

I noticed for example that I do not seem to have observations in which the full board is visible in both camera 3 and camera 4. There are always markers that are hidden in one of the cameras. Is that something necessary for the calibration to work?

closed 2025-08-17 by mprib